leapmotion/leapjs #202: i have put finger-tap-54fps.json.lz in correct  place,but it still can't work.

nanmenxue, 2015-04-17:
it said
XMLHttpRequest cannot load file:///Users/wai121/Desktop/new/finger-tap-54fps.json.lz. Cross origin requests are only supported for protocol schemes: http, data, chrome, chrome-extension, https, chrome-extension-resource.
leap-plugins-0.1.11pre.js:1770 Leap Playback: "finger-tap-54fps.json.lz" seems to be unreachable or the file is empty.
leap-plugins-0.1.11pre.js:1794 Uncaught NetworkError: Failed to execute 'send' on 'XMLHttpRequest': Failed to load 'file:///Users/wai121/Desktop/new/finger-tap-54fps.json.lz'.
leap-0.6.3.min.js:9 Optimized for desktop usage.

but i have placed the finger-tap-54fps.json.lz.

nanmenxue, 2015-04-17:
i use firefox ,it can work! i know why it can't wort.it need http.

pehrlich, 2015-04-17:
You need to host the asset on the same domain.

https://developer.mozilla.org/en-US/docs/Web/HTTP/Access_control_CORS

Like mentioned [here](https://github.com/paperjs/paper.js/issues/673), this is an issue tracker, not a support forum.  These issues are best raised [on the community forum](https://community.leapmotion.com/), where more people will see and be able to get involved.

closed 2015-04-17 by pehrlich