leapmotion/leapjs #203: Accessing Tracking Data from Another Computer

jochasinga, 2015-04-29:
Is there a way that a Node server running leapjs on another computer can connect via websocket to the host computer running leapd on port 6437? i.e. this browser function:

    Leap.loop({host: '192.168.x.xx'}, function(frame) {
        // print frame data to browser console
    });

Of course on browser this didn't work and got net::ERR_CONNECTION_REFUSED, but is it possible to get the data from another node server? (I noticed in the node.js example there's no way to supply the host argument)

capnmidnight, 2015-04-29:
You probably just have your firewall settings blocking that port.

pehrlich, 2015-04-29:
Yes. this is possible.  

Firstly, the code base is almost entirely shared between Node and Web, so passing in the host option should be just the same.

To fix the `net::ERR_CONNECTION_REFUSED`, check out this thread: https://community.leapmotion.com/t/read-leap-data-from-guest-vm-on-host-machine-using-websockets/1523/9

In a nutshell, non-localhost access is by default forbidden, but this can be configured on a per-machine basis.  Also, SSH tunneling is an alternative.

codemercenary, 2015-07-15:
@capnmidnight The webserver is actually configured specifically to accept connections only from localhost.  I'm not exactly sure how I would design a sufficiently secure authentication scheme to serve user input content over the general internet.

tkluysk, 2017-02-26:
For others stumbling to this place two years later...

I had to do a few things:

1. Modify the host IP in base.js (127.0.0.1 by default) to use the IP of the machine the Leap Server is running on.
2. Modify my config.json file as per [this article](https://community.leapmotion.com/t/i-cant-connect-to-leap-device-on-another-pc/1315/2).
3. Disable my local Windows firewall.