leapmotion/leapjs #212: useAnimationLoop() - undefined is not a function

petoem, 2015-08-13:
I get this error: 
```
C:\workspace\private\node_modules\leapjs\lib\controller.js:61
    frameEventName: this.useAnimationLoop() ? 'animationFrame' : 'deviceFrame'
                         ^
TypeError: undefined is not a function
    at Object.module.exports [as Controller] (C:\workspace\private\node_modules\leapjs\lib\controller.js:61:26)
    at Object.<anonymous> (C:\workspace\private\test_on_frame.js:2:23)
    at Module._compile (module.js:460:26)
    at Object.Module._extensions..js (module.js:478:10)
    at Module.load (module.js:355:32)
    at Function.Module._load (module.js:310:12)
    at Function.Module.runMain (module.js:501:10)
    at startup (node.js:129:16)
    at node.js:814:3
```
When I run the following code: 
```javascript
var leap = require('leapjs');
var controller = leap.Controller();

controller.on('frame', function (frame) {
    console.log('LeapFPS: ' + frame.currentFrameRate);
});

controller.connect();
```
Does anyone know how to fix this error or why this happens?

raimo, 2015-09-24:
@petoem Not sure if you figured this out already, but just in case someone else bumps up to this issue, I noticed you're using:

```javascript
    var controller = leap.Controller();
```

whereas the Controller.prototype.useAnimationFrame will be set only if you use it like this:

```javascript
    var controller = new leap.Controller();
```

petoem, 2015-10-05:
@raimo Yes, I've figured it out after reading the "lib/controller.js" code. 
Thanks for your answer :thumbsup:

closed 2015-10-05 by petoem