leapmotion/leapjs #216: Support HMD-mounted orientation, for Orion update?

donmccurdy, 2016-06-01:
Hi, I have a leap device mounted on my HMD, with the Orion update. I'd like to use the leapjs library and THREE.js. The coordinates are all coming out as if the leap were horizontal. Is there, or could we add, a setting like `setOptimizeHMD()` to transform coordinates automatically into HMD-mounted orientation?

Or similarly, would I need to write a plugin that does this? Is it safe/reasonable to modify the frame object dynamically?

pehrlich, 2016-06-01:
It's been a while since anyone's touched it, but the `transform` plugin was designed for exactly this.  Have you seen/tried/discovered any problems with this?

http://leapmotion.github.io/leapjs-plugins/main/transform/

```javascript
 Leap.loop()
    .use('transform', {
      vr: true
    });
```

donmccurdy, 2016-06-01:
That's exactly what I needed, thanks! :smile:

closed 2016-06-01 by donmccurdy