leapmotion/leapjs #217: LeapJS does not work in HTTPS page

sumeetkpatel, 2016-06-04:
Testing the dumper HTTPS seems to try to use a normal web socket instead of a secure one.

Works:
http://leapmotion.github.io/leapjs/examples/dumper.html

Does not work:
https://leapmotion.github.io/leapjs/examples/dumper.html

> WebSocket connection to 'wss://127.0.0.1:6436/v6.json' failed: WebSocket opening handshake was canceled
> Mixed Content: The page at 'https://leapmotion.github.io/leapjs/examples/dumper.html' was loaded over HTTPS, but attempted to connect to the insecure WebSocket endpoint 'ws://127.0.0.1:6437/v6.json'. 
> This request has been blocked; this endpoint must be available over WSS.
> Uncaught SecurityError: Failed to construct 'WebSocket': An insecure WebSocket connection may not be initiated from a page loaded over HTTPS.

pehrlich, 2016-06-06:
Hm.  To troubleshoot your connection, try loading over `https`.  No content will load, but chrome will give you the opportunity to view the certificate: 

https://127.0.0.1:6436/v6.json

[image]

Click details to get:
[image]

You can also try looking in your Registry for a certificate from Leap Motion:

(this is on mac, but the registration details should look the same):
[image]

The registration gets added during installation, so it is possible that a reinstall could fix a missing cert.

sumeetkpatel, 2016-06-07:
Yes, awesome.

Going to that address had an invalid certificate
`There are issues with the site's certificate chain (net::ERR_CERT_DATE_INVALID).`

I accepted the certificate anyway and things worked.
I had only just updated my run-time before posting originally, I guess it didn't refresh the certificate.

closed 2016-06-07 by sumeetkpatel