leapmotion/leapjs #229: Unable to retrieve leap.js for browser

kartik9k, 2019-02-19:
Hey, I wanted to access leap motion library from my browser (chrome) via HTML, js script. I'm attaching the script using : ''' <script src="https://js.leapmotion.com/leap-0.6.4.min.js"></script>
''' but I'm getting an error on my console on the browser '''GET https://js.leapmotion.com/leap-0.6.4.min.js net::ERR_TIMED_OUT'''. Is this a known issue? Am I doing something wrong? Please let me know!

Thanks in advance.

neon-ninja, 2019-12-02:
js.leapmotion.com seems down - unpkg might be a good replacement CDN:
https://unpkg.com/leapjs@0.6.4/leap-0.6.4.min.js

RecursiveCptHook, 2020-04-20:
The https://js.leapmotion.com CDN should be back up and working as expected.

Version 0.6.4 is available at https://js.leapmotion.com/leap-0.6.4.min.js

Version 1.0.0 is not yet available on NPM but is also available at https://js.leapmotion.com/leap-1.0.0.min.js

closed 2020-04-20 by RecursiveCptHook