leapmotion/leapjs #233: leap motion cdn down?

kykungz, 2019-08-09:
I am not able to retrieve any leapjs code from the provided cdn in https://developer-archive.leapmotion.com/javascript

Not even the most important one like https://js.leapmotion.com/leap-0.6.4.min.js

Right now, I can only use leapjs from npm.

theo-armour, 2019-08-09:
try

https://www.jsdelivr.com/package/npm/leapjs

kykungz, 2019-08-11:
@theo-armour Thanks! It works perfectly, but I wonder how could I get leap motion **plugins**?

theo-armour, 2019-08-11:
@kykungz 

Here's an example:

https://cdn.jsdelivr.net/gh/leapmotion/leapjs-plugins@master/main/bone-hand/leap.bone-hand.js

Edit the branch. folder and files to suit your needs.

Learn more at: https://www.jsdelivr.com/ or at https://cdnjs.com/

closed 2019-08-12 by kykungz