leapmotion/leapjs #259: leapjs does not work with Gemini

tstodter, 2022-04-10:
Hello friends, have been struggling with getting my Leap Motion working with the leapjs library (both in the browser and in node.js) with no luck. Downgraded to Orion 4.1.0 and everything's working splendid.

Any suggestions on what might need to be updated to get the javascript up to date with the WebSocket server in the latest tracking software?

jselstad, 2022-07-06:
Unfortunately the latest server does not support LeapJS; a separate application would need to be written to consume LeapC and act as the LeapJS server.

This is an example of a minimal C# layer that consumes the LeapJS schema as a client, so, assuming the server emits these same structures, it should be able to drive v6 LeapJS applications:
https://gist.github.com/jselstad/6c6cbbb558e0b1858baab185eb764960

matejsvajger, 2023-12-11:
There are POC builds of a compatible websocket available for Gemini (WIN/Linux/MacOS(Intel/M Chipset) on their discord server:
https://discord.gg/3VCndThqxS

Tested it successfully with leapjs on MacOS Ventura M2 chip, Ultraleap software 5.17.1 and the very first Leap Motion Controller hardware.